chipweinberger/flutter_blue_plus #1178: [Help]: PlatformException(iosException, -[FlutterStandardTypedData length]: unrecognized selector sent to instance

austinn, 2025-04-17:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.4

### Flutter Version

3.29.3

### What OS?

macOS

### OS Version

15.4.1

### Bluetooth Module

Nordic

### What is your problem?

After upgrade the `flutter_blue_plus` package, I am no longer able to write to the device. I get an internal crash. The app is very bluetooth intensive and has been working fine up until the latest update to `flutter_blue_plus`

Internal crash happens when I call `device.write()`. I haven't changed anything about the data that I am sending

### Logs

```shell
flutter: PlatformException(iosException, -[FlutterStandardTypedData length]: unrecognized selector sent to instance 0x167028ab0, {stackTrace: 0   CoreFoundation                      0x00000001971d5df0 __exceptionPreprocess + 176
1   libobjc.A.dylib                     0x0000000196c9ab60 objc_exception_throw + 88
2   CoreFoundation                      0x0000000197292ce0 -[NSObject(NSObject) __retain_OA] + 0
3   CoreFoundation                      0x0000000197143efc ___forwarding___ + 1500
4   CoreFoundation                      0x0000000197143860 _CF_forwarding_prep_0 + 96
5   flutter_blue_plus_darwin            0x00000001015b6950 -[FlutterBluePlusPlugin handleMethodCall:result:] + 6376
6   FlutterMacOS                        0x000000010729714c __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke + 180
7   FlutterMacOS                        0x00000001065fe2ac -[FlutterEngine engineCallbackOnPlatformMessage:] + 416
8   FlutterMacOS                        0x0000000106e9f6b0 _ZNSt3_fl10__function6__funcIZ23FlutterEngineInitializeE4$_34NS_9allocatorIS2_EEFvNS_10unique_ptrIN7flutter15PlatformMessageENS_14default_deleteIS7_EEEEEEclEOSA_ + 128
9   FlutterMacOS                        0x0000000106eb1f9c _ZN7flutter20PlatformViewEmbedder21HandlePlatformMessageENSt3_fl10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE + 76
10  FlutterMacOS                        0x0000000106eb2ab4 _ZNSt3_fl10__function6__funcIN3fml8internal14CopyableLambdaIZN7flutter20PlatformViewEmbedder30EmbedderPlatformMessageHandler21HandlePlatformMessageENS_10unique_ptrINS5_15PlatformMessageENS_14default_deleteIS9_EEEEEUlvE_EENS_9allocatorISE_EEFvvEEclEv + 80
11  FlutterMacOS                        0x0000000106eaf280 _ZN7flutter18EmbedderTaskRunner8PostTaskEy + 652
12  FlutterMacOS                        0x0000000106e96dcc FlutterEngineRunTask + 36
13  FlutterMacOS                        0x00000001066012cc -[FlutterEngine runTaskOnEmbedder:] + 56
14  FlutterMacOS                        0x0000000106601480 __60-[FlutterEngine postMainThreadTask:targetTimeInNanoseconds:]_block_invoke + 64
15  libdispatch.dylib                   0x0000000196ec1b2c _dispatch_call_block_and_release + 32
16  libdispatch.dylib                   0x0000000196edb85c _dispatch_client_callout + 16
17  libdispatch.dylib                   0x0000000196ef8b2c _dispatch_main_queue_drain.cold.5 + 812
18  libdispatch.dylib                   0x0000000196ed0db0 _dispatch_main_queue_drain + 180
19  libdispatch.dylib                   0x0000000196ed0cec _dispatch_main_queue_callback_4CF + 44
20  CoreFoundation                      0x00000001971a2eb4 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
21  CoreFoundation                      0x0000000197163bac __CFRunLoopRun + 1980
22  CoreFoundation                      0x0000000197162d68 CFRunLoopRunSpecific + 572
23  HIToolbox                           0x00000001a291627c RunCurrentEventLoopInMode + 324
24  HIToolbox                           0x00000001a29194e8 ReceiveNextEventCommon + 676
25  HIToolbox                           0x00000001a2aa4484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
26  AppKit                              0x000000019adb5ab4 _DPSNextEvent + 684
27  AppKit                              0x000000019b754314 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
28  AppKit                              0x000000019ada8c64 -[NSApplication run] + 480
29  AppKit                              0x000000019ad7f35c NSApplicationMain + 880
30  [Dev] UGen PT.debug.dylib           0x00000001020038e4 $sSo21NSApplicationDelegateP6AppKitE4mainyyFZ + 40
31  [Dev] UGen PT.debug.dylib           0x00000001020038ac $s13_Dev__UGen_PT11AppDelegateC5$mainyyFZ + 44
32  [Dev] UGen PT.debug.dylib           0x000000010200394c __debug_main_executable_dylib_entry_point + 28
33  dyld                                0x0000000196cdab4c start + 6000}, null)
```

liaowei12138, 2025-04-18:
You can try to replace this file.

/Users/YourName/.pub-cache/hosted/pub.dev/flutter_blue_plus_darwin-4.0.0/darwin/flutter_blue_plus_darwin/Sources/flutter_blue_plus_darwin/FlutterBluePlusPlugin.m

The new file's download url：https://drive.google.com/file/d/1IJQz6XFuu6vY4I8eGxlGmZRXtW_3XAte/view?usp=drive_link

chipweinberger, 2025-04-18:
https://github.com/chipweinberger/flutter_blue_plus/issues/1178

closed 2025-04-18 by chipweinberger

bhavesh379, 2025-04-18:
How this get closed, I am facing the same issue and within few hours it get solved without any comments, can you please help me regarding this @chipweinberger @austinn 

I am not able to get value from onValueReceived in iOS but there I am able to get correct values from android.

exception I am getting as below-
PlatformException(iosException, -[FlutterStandardTypedData length]: unrecognized selector sent to instance 0x14c6dd200, {stackTrace: 0   CoreFoundation                      0x00000001983152f8 0551BAA3-C6E9-3423-B0E2-BE69719C1D7F + 1155832
1   libobjc.A.dylib                     0x0000000195799a7c objc_exception_throw + 88
2   CoreFoundation                      0x000000019837f67c 0551BAA3-C6E9-3423-B0E2-BE69719C1D7F + 1590908
3   CoreFoundation                      0x000000019822cb84 0551BAA3-C6E9-3423-B0E2-BE69719C1D7F + 203652
4   CoreFoundation                      0x000000019822c4f0 _CF_forwarding_prep_0 + 96
5   flutter_blue_plus_darwin            0x00000001056c04d0 -[FlutterBluePlusPlugin handleMethodCall:result:] + 16564
6   Flutter                             0x000000010f6d962c __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke + 164
7   Flutter                             0x000000010f155574 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt3_fl10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116
8   libdispatch.dylib                   0x0000000106a0c584 _dispatch_call_block_and_release + 32
9   libdispatch.dylib                   0x0000000106a26064 _dispatch_client_callout + 16
10  libdispatch.dylib                   0x0000000106a46f38 _dispatch_main_queue_drain.cold.5 + 876
11  libdispatch.dylib                   0x0000000106a1c548 _dispatch_main_queue_drain + 180
12  libdispatch.dylib                   0x0000000106a1c484 _dispatch_main_queue_callback_4CF + 44
13  CoreFoundation                      0x000000019826e2b4 0551BAA3-C6E9-3423-B0E2-BE69719C1D7F + 471732
14  CoreFoundation                      0x000000019826c0b0 0551BAA3-C6E9-3423-B0E2-BE69719C1D7F + 463024
15  CoreFoundation                      0x0000000198290700 CFRunLoopRunSpecific + 572
16  GraphicsServices                    0x00000001e4dd1190 GSEventRunModal + 168
17  UIKitCore                           0x000000019aeae240 05197341-F998-32E5-A847-DA98BB8C68BF + 3973696
18  UIKitCore                           0x000000019aeac470 UIApplicationMain + 336
19  UIKitCore                           0x000000019b30aa30 05197341-F998-32E5-A847-DA98BB8C68BF + 8546864
20  Runner.debug.dylib                  0x0000000105aa8ddc $sSo21UIApplicationDelegateP5UIKitE4mainyyFZ + 128
21  Runner.debug.dylib                  0x0000000105aa8d4c $s6Runner11AppDelegateC5$mainyyFZ + 44
22  Runner.debug.dylib                  0x0000000105aa8e58 __debug_main_executable_dylib_entry_point + 28
23  dyld                                0x00000001bec93ad8 189FE480-5D5B-3B89-9289-58BC88624420 + 68312}, null)

austinn, 2025-04-18:
> [#1178](https://github.com/chipweinberger/flutter_blue_plus/issues/1178)

@chipweinberger Why was this closed and then referenced this issue?

ekuleshov, 2025-04-19:
@austinn in my understanding the error is suspected to be due to mismatched versions of platform-specific plugins.

Please check the example app. If example app works on your devices, please share content of your `pubspec.yaml`

chipweinberger, 2025-04-20:
sorry meant this: https://github.com/chipweinberger/flutter_blue_plus/issues/1163#issuecomment-2816506628