chipweinberger/flutter_blue_plus #1180: [Help]:  if(manufData != null)  变量 manufData位置:类 FlutterBluePlusPlugin FAILURE: Build failed with an exception.

dxr199212300, 2025-04-18:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

flutter_blue_plus: ^1.14.11

### Flutter Version

3.22.0

### What OS?

Android

### OS Version

android

### Bluetooth Module

三星 S21

### What is your problem?

flutter run build error

### Logs

```shell
flutter_blue_plus_android-4.0.0/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2611: 错误: 找不到符号
        if (manufData != null) {
            ^
  符号:   变量 manufData
  位置: 类 FlutterBluePlusPlugin
```

swat199538, 2025-04-18:
me too.
我发现我flutter pub add  shared_preferences 依赖的flutter_blue_plus_android 就会自动升级到 4.0我把我的依赖改成
flutter_blue_plus: 1.35.2 去掉了 ^ 就没有升级到4.0暂时屏蔽掉了错误。

rFronteddu, 2025-04-18:
I get the same error

MrCsabaToth, 2025-04-19:
Duplicate of #1177

closed 2025-04-21 by chipweinberger

tangchi-looki, 2025-04-21:
[image]

Launching lib/main.dart on Pixel 8 in debug mode...
/Users/edy/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.1/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:1614: 警告: [deprecation] Secure中的LOCATION_MODE已过时
                Settings.Secure.LOCATION_MODE, Settings.Secure.LOCATION_MODE_OFF);
                               ^
/Users/edy/.pub-cache/hosted/pub.dev/flutter_blue_plus_android-4.0.1/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java:2651: 错误: 找不到符号
        if (manufData != null)           {map.put("manufacturer_data", manufDataB);}
            ^
  符号:   变量 manufData
  位置: 类 FlutterBluePlusPlugin
1 个错误
1 个警告

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus_android:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --info option to get more log output.
> Run with --scan to get full insights.

BUILD FAILED in 4s
Error: Gradle task assembleDebug failed with exit code 1

Exited (1).

@MrCsabaToth

MrCsabaToth, 2025-04-21:
Missing `manufData` is #1177, what is your `flutter_blue_plus_android` Android version? Either pin it back or there'll be a release soon with the recent fix.

tangchi-looki, 2025-04-21:
> Missing `manufData` is [#1177](https://github.com/chipweinberger/flutter_blue_plus/issues/1177), what is your `flutter_blue_plus_android` Android version? Either pin it back or there'll be a release soon with the recent fix.

I've upgraded from 4.0.1 to 4.0.2, and the issue has been resolved.