chipweinberger/flutter_blue_plus #1181: [Help]: ManufacturedData is Always empty on iOS

j20mc, 2025-04-18:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.3

### Flutter Version

3.29

### What OS?

iOS

### OS Version

16.1.1

### Bluetooth Module

BlueUP Forte+ BLE

### What is your problem?

Hello, 

I have make same test from android and iOS, in iOS, I have a ManufacturedData empty

`manufacturerData: {}`

My config : 

`[✓] Flutter (Channel stable, 3.29.0, on macOS 15.3.2 24D81
    darwin-arm64, locale fr-FR) [1 066ms]
    • Flutter version 3.29.0 on channel stable at
      /Users/jmc/Development/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 35c388afb5 (10 weeks ago), 2025-02-10
      12:48:41 -0800
    • Engine revision f73bfc4522
    • Dart version 3.7.0
    • DevTools version 2.42.2

[✓] Android toolchain - develop for Android devices (Android SDK
    version 35.0.0-rc3) [1 396ms]
    • Android SDK at /Users/jmc/Library/Android/sdk
    • Platform android-35, build-tools 35.0.0-rc3
    • Java binary at: /Applications/Android
      Studio.app/Contents/jbr/Contents/Home/bin/java
      This is the JDK bundled with the latest Android Studio
      installation on this machine.
      To manually set the JDK path, use: `flutter config
      --jdk-dir="path/to/jdk"`.
    • Java version OpenJDK Runtime Environment (build
      17.0.10+0-17.0.10b1087.21-11572160)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 16.2) [792ms]
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Build 16C5032a
    • CocoaPods version 1.16.2

[✓] Chrome - develop for the web [9ms]
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google
      Chrome

[✓] Android Studio (version 2023.3) [9ms]
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build
      17.0.10+0-17.0.10b1087.21-11572160)`

### Logs

```shell
flutter: ScanResult{device: BluetoothDevice{remoteId: 01DA57B4-BE9F-4957-F554-FB02B0EF63D7, platformName: BlueUp-04-077578, services: null}, advertisementData: AdvertisementData{advName: BlueUp-04-077578, txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {}, serviceData: {8800: [66], 180f: [100]}, serviceUuids: [a3c87500-8ed3-4bdf-8a39-a01bebede295]}, rssi: -62, timeStamp: 2025-04-18 15:31:11.699771}
flutter: ScanResult{device: BluetoothDevice{remoteId: E72B32E0-1F64-C0DD-6C78-845268324F12, platformName: BlueUp-04-077577, services: null}, advertisementData: AdvertisementData{advName: BlueUp-04-077577, txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {}, serviceData: {8800: [66], 180f: [100]}, serviceUuids: [a3c87500-8ed3-4bdf-8a39-a01bebede295]}, rssi: -58, timeStamp: 2025-04-18 15:31:08.796333}
```

chipweinberger, 2025-04-21:
works fine for me

closed 2025-04-21 by chipweinberger

j20mc, 2025-04-21:
Ok, it's works fine for you, but I have a problem. Logically, this plugin gives a same information iOS & Android in ManufacturedData. I my case not.

Can I ask you what versions you have tried on (Flutter / flutter_blue_plus / ios), might this solve my problem?

chipweinberger, 2025-04-21:
try updating

j20mc, 2025-04-28:
Hello y try with this update :

flutter_blue_plus: ^1.35.4
Flutter 3.29.3

iOS →

`BluetoothDevice{remoteId: 38821DB7-8216-E214-80F1-A820344A6930, platformName: BlueUp-04-077579, services: null}, advertisementData: AdvertisementData{advName: BlueUp-04-077579, txPowerLevel: null, appearance: null, connectable: true, **manufacturerData: {}**, serviceData: {8800: [66], 180f: [99]}, serviceUuids: [a3c87500-8ed3-4bdf-8a39-a01bebede295]}, rssi: -76, timeStamp: 2025-04-28 10:10:54.448403}`

Android →

`BluetoothDevice{remoteId: C1:03:59:88:60:24, platformName: BlueUp-04-077577, services: null}, advertisementData: AdvertisementData{advName: , txPowerLevel: null, appearance: null, connectable: false, **manufacturerData: {76: [2, 21, 78, 69, 79, 83, 65, 70, 69, 45, 48, 50, 48, 49, 55, 49, 54, 48, 1, 151, 29, 153, 194]}**, serviceData: {}, serviceUuids: []}, rssi: -71, timeStamp: 2025-04-28 10:20:45.533880}`

But manufacturerData is alwals empty