chipweinberger/flutter_blue_plus #1189: [Help]: Should the library crash in Linux when listening for `FlutterBluePlus.adapterState` if the Bluetooth is unavailable?

tijsmoree, 2025-04-24:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.3

### Flutter Version

3.29.2

### What OS?

All

### OS Version

Ubuntu 24.04.2

### Bluetooth Module

Intel Corp. AX201 Bluetooth

### What is your problem?

When listening for `FlutterBluePlus.adapterState` it gives a `Bad State: No element` on Linux when my Bluetooth is disabled. I can fix it by checking `FlutterBluePlus.isSupported` first, but still, should it crash?

It happens because `flutter_blue_plus_linux.dart:446` uses `first` on the empty list `_client.adapters`, I believe.

### Logs

```shell
Launching lib/main.dart on Linux in debug mode...
✓ Built build/linux/x64/debug/bundle/adepth
Connecting to VM Service at ws://127.0.0.1:45717/GcZ0hhgL2Dg=/ws
Connected to the VM Service.

════════ Exception caught by provider ══════════════════════════════════════════
The following assertion was thrown:
An exception was throw by _ControllerStream<BluetoothAdapterState> listened by
StreamProvider<BluetoothAdapterState>, but no `catchError` was provided.

Exception:
Bad state: No element

════════════════════════════════════════════════════════════════════════════════
```

chipweinberger, 2025-04-25:
@tnc1997

chipweinberger, 2025-04-25:
throw exception != crash

tijsmoree, 2025-04-25:
Ah, I now see that `flutter_blue_plus_linux` v3.0.1 already solves this issue. I was still on v3.0.0.

PR #1190 (2025-04-25): [Refactor] Return unknown state if adaptors list empty
Closes #1189
closed 2025-04-25 by tnc1997