chipweinberger/flutter_blue_plus #1192: [Help]: Location services are required for Bluetooth scan even on Android 15 (API 35)

keval-devani, 2025-04-25:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.4

### Flutter Version

3.29.3

### What OS?

macOS

### OS Version

15.3.2

### Bluetooth Module

Scanning for BLE device

### What is your problem?

t looks like we misplaced the isLocationEnabled check. If the app doesn't intend to use location for scanning, there's no need to check for it at all. This unnecessary check was introduced in https://github.com/chipweinberger/flutter_blue_plus/issues/1167 [code](https://github.com/chipweinberger/flutter_blue_plus/blob/master/packages/flutter_blue_plus_android/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L504)

According to the [FlutterBluePlus documentation](https://pub.dev/packages/flutter_blue_plus#add-permissions-for-android-no-location), location permissions (and checks like isLocationEnabled) are only required if the app actually uses location-based scanning

### Logs

```shell
PlatformException(startScan, Location services are required for Bluetooth scan, null, null)
```

ekuleshov, 2025-04-25:
@keval-devani this is not a location permission check but a check for system location services.

I can't check on Android 15 right now, but on 10..12 the location services are required even when bluetooth scan permission. Even if you give permission the BLE scan returns nothing with location services disabled.

keval-devani, 2025-04-25:
@ekuleshov I tested version `flutter_blue_plus: 1.35.3` Android was scanning and returning results on android 14 and now on android 15 that I can verify without **location service turn on**. 

Please see here https://developer.android.com/develop/connectivity/bluetooth/bt-permissions

keval-devani, 2025-04-25:
@ekuleshov Yes I agree with your response on article. 

Right now I'm checking on Android 9 (API Level 28) using `flutter_blue_plus: 1.35.3` with location service is turn off and getting scanResults 🤨

William-Geng, 2025-04-26:
Can confirm! when using 1.35.4 with flutter_blue_plus_android 4.0.2, on android 14 and 15 you would get this error if you try to start scan with location service off.

Downgrading to an older version like I.34.5 would solve the issue - did anything change?

chipweinberger, 2025-04-26:
**Location Services** are required for scanning. **Location Permissions** are _not_ required.

chipweinberger, 2025-04-26:
closing.

If you want us to change this behavior, please open a new issue with a correct title.

permissions != services

ekuleshov, 2025-04-26:
@chipweinberger I poked around. I thought that _location services_ might be also tied to the `BLUETOOTH_SCAN` permission [introduced for Android 12](https://developer.android.com/about/versions/12/features/bluetooth-permissions). 

Though I could not find any reference in the Android API documentation for that.

Also did some tests:

* Android 5.1.1 (Amazon Fire 7) - the scan with location services disabled works (this version is before runtime user permissions introduced, i.e. before Android 6)

* Android 11 (Amazon Fire 8) - the scan with location services disabled doesn't return any results

* Android 12 (Google Pixel 3) - the scan with location services disabled works

* Android 14 (Samsung) - the scan with location services disabled works

@chipweinberger so, I'm thinking of disabling the _location services_ check for Android 12+ and perhaps also add a flag like 
`androidCheckLocationServices` to allow disable it for the rest (`true` by default).

ekuleshov, 2025-04-27:
Starting from 4.0.3 version the Android plugin does not check for location services on Android 12+ and you can explicitly disable location services check in the startScan() call parameters.

You can run `flutter pub upgrade`

closed 2025-04-27 by ekuleshov

shingohu, 2025-05-09:
> [@chipweinberger](https://github.com/chipweinberger) I poked around. I thought that _location services_ might be also tied to the `BLUETOOTH_SCAN` permission [introduced for Android [...]

Android6.0-11 should enable location services ,and Android 10 & 11 need ACCESS_FINE_LOCATION permission, below Android10 need ACCESS_COARSE_LOCATION permission