chipweinberger/flutter_blue_plus #1203: [Help]: Failed to write byte: -[FlutterStandardTypedData length]: unrecognized selector sent to instance 0x13388xxx

mastrs, 2025-05-04:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.35.4

### Flutter Version

3.29.3

### What OS?

iOS

### OS Version

18.4.1

### Bluetooth Module

Silicon Labs EFR32XG22

### What is your problem?

For any iOS write attempt on remote device, including via the example app, I get following error:
`flutter: Failed to write byte: -[FlutterStandardTypedData length]: unrecognized selector sent to instance 0x133882480`

### Logs

```shell
Failed to write byte: -[FlutterStandardTypedData length]: unrecognized selector sent to instance 0x133882480
```

mastrs, 2025-05-04:
This fixed it for me (FlutterBLuPlusPlugin.m):
```
 else if ([@"writeCharacteristic" isEqualToString:call.method])
        {
            // See BmWriteCharacteristicRequest
            NSDictionary *args = (NSDictionary*)call.arguments;
            NSString  *remoteId           = args[@"remote_id"];
            NSString  *serviceUuid        = args[@"service_uuid"];
            NSString  *characteristicUuid = args[@"characteristic_uuid"];
            NSString  *primaryServiceUuid = args[@"primary_service_uuid"];
            NSNumber  *writeTypeNumber    = args[@"write_type"];
            NSNumber  *allowLongWrite     = args[@"allow_long_write"];
            NSData    *value              = args[@"value"];

 NSData *dataBytes;

            if ([value isKindOfClass:[FlutterStandardTypedData class]]) {
              // Dart sent a Uint8List / ByteBuffer
                FlutterStandardTypedData *fstd = value;
                dataBytes = fstd.data; //[(FlutterStandardTypedData*)value data];
                NSLog(@"✅ Data FlutterStandard rawValueType=%@", dataBytes);
                value = dataBytes;
            } 
```

ekuleshov, 2025-05-04:
@mastrs please open a PR. Make sure you don't modify code outside of your fix.
Also please provide some more detailed steps how to reproduce issue.

chipweinberger, 2025-05-04:
upgrade.

closed 2025-05-04 by chipweinberger