chipweinberger/flutter_blue_plus #1206: [Feature]: Don't limit RxDart version

chipweinberger, 2025-05-06:
### FlutterBluePlus Version

1.35.X

### Flutter Version

N/A

### What OS?

All

### OS Version

xx

### Bluetooth Module

xxx

### What is your feature request?

Currently using FBP requires `rxdart: '>=0.27.0 <0.29.0'`

This internal detail should not be leaking into our public facing API. Users should be able to use the rxdart of their choosing.

We need to either:

1. copy the rxdart souce into FBP-linux 
2. remove rxdart as a dependency
3. find a way to let users choose their own version of rxdart (I dont think this is possible)

### Logs

```shell
xxx
```

chipweinberger, 2025-05-06:
@tnc1997

tnc1997, 2025-05-07:
> limiting users a specific range of versions for RxDart

0.27.0 was released on 7 May 2021 and 0.28.0 is the latest version therefore unless someone is using an older version this is no problem.

> this is annoying

Out of interest have any developers raised the version constraint as an issue?

> copy the rxdart souce into FBP-linux

I would recommend against this personally as it increases maintenance overhead and we would need our own unit tests to validate the logic.

chipweinberger, 2025-05-07:
> Out of interest have any developers raised the version constraint as an issue?

yes. here's an older example, https://github.com/espresso3389/pdfrx/issues/211 RxDart is still actively developed.

> I would recommend against this personally as it increases maintenance overhead and we would need our own unit tests to validate the logic.

how would it increase maintenance? As long as the code works, we don't need to "maintain" it. Why would it break?

chipweinberger, 2025-05-07:
To be clear, I would most strongly advocate for this option: 

> remove rxdart as a dependency

A small utility function should be able to replace the need for it.

closed 2025-05-07 by chipweinberger

tnc1997, 2025-05-07:
> yes. here are some examples.

I meant developers raising issues with `flutter_blue_plus` specifically.

> internal code decisions should not be part of the public facing API whenever possible.

None of the internals of `rxdart` are exposed via the [public facing API](https://github.com/chipweinberger/flutter_blue_plus/blob/master/packages/flutter_blue_plus/lib/flutter_blue_plus.dart).

chipweinberger, 2025-05-07:
Why do you want to keep RxDart?

tnc1997, 2025-05-07:
> Why do you want to keep RxDart?

Because many of its classes and functions are used in the Linux implementation and it avoids reinventing the wheel.

chipweinberger, 2025-05-07:
hmm.

I only see two uses? 

* MergeStream (which we *already* have defined elsewhere)
* switchMap

is there something else im missing?

And here is swtichMap in vanilla dart. Do we really need a whole big library for this?

```
extension SwitchMapExtension<T> on Stream<T> {
  /// For each event from the source stream, cancels any previous inner
  /// subscription and listens to the stream returned by [mapper].
  /// Completes only after the source has finished *and* the current inner
  /// stream (if any) has finished.
  Stream<S> switchMap<S>(Stream<S> Function(T value) mapper) {
    // Controller for the output
    final controller = StreamController<S>();
    StreamSubscription<T>? outerSub;
    StreamSubscription<S>? innerSub;
    var outerDone = false;

    // Helper to close once both outer is done and no inner is active
    void tryClose() {
      if (outerDone && innerSub == null) {
        controller.close();
      }
    }

    controller
      // Start listening to source when someone listens to the result
      ..onListen = () {
        outerSub = this.listen(
          (value) {
            // Cancel previous inner
            innerSub?.cancel();
            // Listen to the new inner stream
            innerSub = mapper(value).listen(
              controller.add,
              onError: controller.addError,
              onDone: () {
                innerSub = null;
                tryClose();
              },
            );
          },
          onError: controller.addError,
          onDone: () {
            outerDone = true;
            tryClose();
          },
        );
      }
      // Propagate pause/resume to both subscriptions
      ..onPause = () {
        outerSub?.pause();
        innerSub?.pause();
      }
      ..onResume = () {
        outerSub?.resume();
        innerSub?.resume();
      }
      // Cancel both if the listener cancels
      ..onCancel = () async {
        await outerSub?.cancel();
        await innerSub?.cancel();
      };

    return controller.stream;
  }
}
```

tnc1997, 2025-05-07:
I believe that it is better to reuse common well-tested implementations than reinvent the wheel and bundle your own copy/pasted equivalents, but of course you are welcome to remove the dependency and maintain your own if you would prefer. There are pros and cons to both approaches some of which are discussed in [this](https://news.ycombinator.com/item?id=11403005) thread and [this](https://news.ycombinator.com/item?id=14903059) thread.

chipweinberger, 2025-05-07:
> I believe that it is better to reuse common well-tested implementations than reinvent the wheel and bundle your own copy/pasted equivalents

Why? Copy pasting takes two seconds. Adding a dependency limits who can use FBP. Version resolution is not a simple task in a big app. Thats a huge tradeoff to save two seconds.

Personally, I'd rather remove the linux support than leak RxDart. If RxDart was not popular it might be okay. But RxDart is very popular and can be hard to resolve. This is why there are so many Github issues about it.

tnc1997, 2025-05-07:
> if we update FBP to support the newer RxDart now users will be stuck on the latest version

It is worth noting that our version constraint means that developers can use any version since 0.27.0 (7 May 2021).

This means that developers are not stuck with the latest version of `rxdart` and can use any compatible version.

chipweinberger, 2025-05-07:
1. new RxDart 30.0 comes out
2. other package requires 30.0 for whatever reason
3. FBP does not list support for 30.0
4. we update FBP
5. user is now stuck on latest FBP or must downgrade other packages too

tnc1997, 2025-05-07:
> 5\. user is now stuck on latest FBP

If other packages are being updated to support 30.0 then including FBP in those is not unreasonable in my opinion.

chipweinberger, 2025-05-07:
not unreasonable to be forced to upgrade because of a single util function we didnt want to spend 2 seconds copy pasting?

tnc1997, 2025-05-07:
> RxDart is very popular

The popularity of RxDart as a well maintained library is one of the reasons to use it instead of copy/pasting its internals.

> we didnt want to spend 2 seconds copy pasting

Like I have previously said above you are welcome to remove the dependency and maintain your own implementations.

chipweinberger, 2025-05-10:
Just to add to the discussion. Users do worry about dependencies. 

[Rust’s dependencies are starting to worry me](https://vincents.dev/blog/rust-dependencies-scare-me/?)
https://news.ycombinator.com/item?id=43935067