chipweinberger/flutter_blue_plus #1210: [Help]: 开始扫描后，FlutterBluePlus.onScanResults回调有几率不返回任何结果，但FlutterBluePlus.isScanning显示扫描仍然在进行，直到超时变成false

godfeather, 2025-05-09:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.5

### Flutter Version

3.29.1

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Bluetooth 5.3

### What is your problem?

FlutterBluePlus.onScanResults.listen有概率只调用一次，并且日志中会出现一条：/BluetoothLeScanner( 3754): onScannerRegistered() - status=6 scannerId=-1 mScannerId=0，此时回调再也不会触发，但FlutterBluePlus.isScanning却一直持续到超时变为false

以下是核心代码：

[image]

### Logs

```shell
I/flutter ( 4721): scan
I/flutter ( 4721): scaning: 0
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
W/WindowOnBackDispatcher( 4721): OnBackInvokedCallback is not enabled for the application.
W/WindowOnBackDispatcher( 4721): Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
D/[FBP-Android]( 4721): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 4721): isLeEnabled(): ON
D/BluetoothLeScanner( 4721): onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
I/flutter ( 4721): scaning: 0
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
E/err     ( 4721): 所有权限已经授权！
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
E/BLASTBufferQueue( 4721): [SurfaceView[com.raymoniot.solarhub2/com.raymoniot.solarhub2.MainActivity]#1](f:1,a:4) acquireNextBufferLocked: Can't acquire next buffer. Already acquired max frames 4 max:2 + 2
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
E/BLASTBufferQueue( 4721): [SurfaceView[com.raymoniot.solarhub2/com.raymoniot.solarhub2.MainActivity]#1](f:1,a:4) acquireNextBufferLocked: Can't acquire next buffer. Already acquired max frames 4 max:2 + 2
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
I/oniot.solarhub2( 4721): This is sticky GC, maxfree is 8388608 minfree is 2097152
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
D/WIFI STATUS( 4721): WIFI状态: 3
I/System.out( 4721): 非双频
I/System.out( 4721): WIFI状态：3
I/flutter ( 4721): 扫描结束
```

chipweinberger, 2025-05-09:
so you're saying that when a scan fails, `FlutterBluePlus.isScanning` still reports `true` until the timeout? 

Yes that seems like a bug. please open a PR.

chipweinberger, 2025-05-09:
However, the code looks fine. If we fail, we call `_stopScan` which updates `isScanning`.

```
      // invoke platform method
      await _invokeMethod(() => FlutterBluePlusPlatform.instance.startScan(settings)).onError((e, s) {
        _stopScan(invokePlatform: false);
        throw e!;
      });
```

godfeather, 2025-05-12:
Hi engineer, please try this way:

bluescan function:
[image]

loop call the 'bluescan' function: 

[image]

Problem：

[image]

Logs：
[bluescan.log](https://github.com/user-attachments/files/20153399/bluescan.log)

Source Code: 

[bluescan.txt](https://github.com/user-attachments/files/20153493/bluescan.txt)

Thanks in Advance

chipweinberger, 2025-05-12:
hmm, i think android is failing to report the error to us. maybe it's an android issue.

chipweinberger, 2025-05-22:
code looks fine

closed 2025-05-22 by chipweinberger