chipweinberger/flutter_blue_plus #1216: [Help]: Partial Name Matching During Scanning

shijianoo, 2025-05-29:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.5

### Flutter Version

3.32.0

### What OS?

Android

### OS Version

13

### Bluetooth Module

unkonw

### What is your problem?

I'm currently working on a project that needs to scan for Bluetooth LE devices whose names contain a specific substring (not just exact matches). I noticed that the startScan() method accepts a withNames parameter, but it only supports exact matches.

As a result, I'm currently left with filtering the results manually like this:
```
FlutterBluePlus.scanResults.listen((results) {
  for (ScanResult r in results) {
    if (r.device.name.contains("TargetSubstring")) {
      // Process device
    }
  }
});
```

Is there a better or more efficient way to handle this use case? For example:

A built-in nameContains or pattern matching option in startScan()

Or perhaps a suggested way to avoid filtering manually in the scanResults.listen stream?

### Logs

```shell
No log required
```

ekuleshov, 2025-05-29:
The `withNames` and other parameters are going to the native-platform filter used for scanning. Those filters don't have support for partial name matching. So, you have to do your own filtering of the scan results and the only improvement you could do is to use some other filter options to narrow down your scan results, e.g. us your devices advertise some service uuids.

chipweinberger, 2025-06-04:
or use withKeywords. a bit better