chipweinberger/flutter_blue_plus #1225: [Help]: App crash and UI lag on Android (Galaxy M04) after Flutter upgrade with continuous BLE notifications

mrunal-atreya, 2025-06-20:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

  flutter_blue_plus: ^1.35.5

### Flutter Version

3.32.4

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

I am using a Samsung Galaxy M04. I am not sure about the exact BLE module, but can try to find out if needed.

### What is your problem?

**Description:**
After upgrading Flutter from 3.27 to the latest stable version, my app (using flutter_blue_plus for BLE notifications) started experiencing severe UI lag and eventually crashes on some Android devices (notably Galaxy M04).
This issue does not occur on iOS or on other Android devices (e.g., Poco).
Reverting to an APK built with the older Flutter version resolves the problem, even with the same plugin version.

**To Reproduce:**
Use flutter_blue_plus to subscribe to a BLE characteristic that streams data continuously (e.g., for a live graph).
Build and run the app on a Galaxy M04 (Android 13).
Observe the graph updating in real time.
After a few seconds/minutes, the UI becomes increasingly laggy and the app eventually crashes

**Expected behavior:**
The app should handle continuous BLE notifications smoothly, as it did before the Flutter upgrade.

**Code Sample:**
`// Subscribe to BLE notifications and update graph in real time
Future<void> readData() async {
  await characteristic!.setNotifyValue(true);
  sub = characteristic!.lastValueStream.listen((value) {
    if (value.isNotEmpty) {
      allRawData.addAll(value);
      // Parse and process BLE data
      var sensorData = getGraphData(value, categoryList1.length, categoryList2.length, categoryList3.length);
      if (sensorData[0].isNotEmpty && sensorData[1].isNotEmpty && sensorData[2].isNotEmpty) {
        categoryList1.addAll(sensorData[0]);
        categoryList2.addAll(sensorData[1]);
        categoryList3.addAll(sensorData[2]);
        // Update graph every second
        if (previousSecond != seconds) {
          previousSecond = seconds;
          _buildLineChart(
            categoryList1.length > 2000 ? categoryList1.sublist(categoryList1.length - 2000) : categoryList1,
            categoryList2.length > 2000 ? categoryList2.sublist(categoryList2.length - 2000) : categoryList2,
            categoryList3.length > 2000 ? categoryList3.sublist(categoryList3.length - 2000) : categoryList3,
            updatePressureIndicator(categoryList1, categoryList2, categoryList3),
          );
        }
      }
      setState(() {}); // UI update
    }
[... 991 lines cut ...]
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:
D/[FBP-Android](11582): [FBP]   chr: ffe1
D/[FBP-Android](11582): [FBP] onCharacteristicChanged:

Lost connection to device.

Exited.
```

mrunal-atreya, 2025-06-20:
@chipweinberger @tnc1997 @boskokg @ekuleshov @mbalamut @officelioneight @MrCsabaToth @tuna-f1sh @Zyr00 @panmari @wcoder @danielle-h 
Could someone please look into this issue on high priority? It’s a critical blocker for us — our entire app functionality relies on the flutter_blue_plus package, and due to this issue, doctors are currently unable to use the app. We would really appreciate a quick resolution.

chipweinberger, 2025-06-25:
please dont spam so many people

Allhad02, 2025-06-26:
We've resolved this issue by downgrading Flutter to version 3.27, where the app behaves as expected — no crashes, smooth UI updates, and consistent BLE data handling, even on low-end devices like the Samsung Galaxy M04.

**For Others Facing This Issue:**
After upgrading to Flutter 3.32.x, our app began showing the following symptoms:
1. Severe UI lag after a few seconds runtime
2. App crash without meaningful error logs (especially on devices with limited hardware resources)
3. Issue reproducible only on some Android devices (like Galaxy M04), not on iOS or higher-end Android devices

**Root Cause (based on investigation):**
Flutter 3.32 introduced significant changes in the rendering and memory management system:
1. Upgraded Skia engine and Dart GC
2. Stricter memory and rendering pipeline behavior
3. Changes in native heap management that now crash apps which previously ran near the memory limit
4. This particularly affects apps dealing with real-time data streams, graph updates

**In our case:**
1. We were subscribing to a high-frequency BLE notification stream
2. App processed and appended data to 3 large lists 
3. We were plotting live graphs by feeding large data points to the UI repeatedly
4. This caused increasing memory usage and eventually led to crashes under the stricter environment of Flutter 3.32

**Recommended Actions**
If you're facing similar issues after upgrading:
Option 1: Temporary Fix
Downgrade Flutter to 3.27
1. This version handles memory and rendering more leniently

Option 2: Long-term Optimization (Recommended)
1. Throttle BLE data updates.
2. Maintain a rolling data window .
3. Avoid calling setState() on every BLE packet
4. Use WidgetsBinding.instance.addPostFrameCallback to defer UI updates and reduce UI thread congestion
5. Consider offloading heavy processing to isolates if needed