chipweinberger/flutter_blue_plus #1227: [Help]:

shingohu, 2025-07-07:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.5

### Flutter Version

3.32.5

### What OS?

Android

### OS Version

Android 12/13/14 

### Bluetooth Module

HC

### What is your problem?

HotRestart can cause multiple callbacks to be executed for Bluetooth connection status

1、connect a ble device(autoconnect is true)
2、ble device poweroff and will receive disconnect state callback
3、hotRestart
4、ble device poweron and will receive multiple connect state callback

I tried to perform the same operation on mAutoConnected as mConnectedDevices in the disconnectAllDevices method, and the problem was solved. So should we perform the same operation on mCurrently ConnectingDevices mBondingDevices and mAutoConnected as mConnectedDevices?

### Logs

```shell
2025-07-07 15:49:31.460 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onConnectionStateChange:connected
2025-07-07 15:49:31.460 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: SUCCESS
2025-07-07 15:49:31.480 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: discoverServices
2025-07-07 15:49:33.345 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onServicesDiscovered:
2025-07-07 15:49:33.346 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   count: 5
2025-07-07 15:49:33.346 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: 0GATT_SUCCESS
2025-07-07 15:49:33.413 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: setNotifyValue
2025-07-07 15:49:33.424 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: disconnect
2025-07-07 15:49:33.434 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onConnectionStateChange:disconnected
2025-07-07 15:49:33.434 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: SUCCESS
2025-07-07 15:49:33.478 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: startScan
2025-07-07 15:49:33.621 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: stopScan
2025-07-07 15:49:33.637 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: connect
2025-07-07 15:49:33.895 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP] onConnectionStateChange:connected
2025-07-07 15:49:33.895 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: SUCCESS
2025-07-07 15:49:33.897 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: discoverServices
2025-07-07 15:49:35.819 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP] onServicesDiscovered:
2025-07-07 15:49:35.819 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   count: 5
2025-07-07 15:49:35.819 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: 0GATT_SUCCESS
2025-07-07 15:49:35.831 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: setNotifyValue
2025-07-07 15:49:35.914 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP] onDescriptorWrite:
2025-07-07 15:49:35.914 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   chr: 2a05
2025-07-07 15:49:35.914 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   desc: 2902
2025-07-07 15:49:35.914 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:49:35.924 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: setNotifyValue
2025-07-07 15:49:36.015 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP] onDescriptorWrite:
2025-07-07 15:49:36.015 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   chr: 4102
2025-07-07 15:49:36.015 20579-21125 [FBP-Android]           com.firo.mengqin                     D  [FBP]   desc: 2902
[... 71 lines cut ...]
2025-07-07 15:50:45.221 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   desc: 2902
2025-07-07 15:50:45.221 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:50:45.579 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: requestMtu
2025-07-07 15:50:45.597 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMtuChanged:
2025-07-07 15:50:45.597 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   mtu: 512
2025-07-07 15:50:45.597 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:50:45.614 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: writeCharacteristic
2025-07-07 15:50:45.623 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onCharacteristicWrite:
2025-07-07 15:50:45.624 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   chr: 5101
2025-07-07 15:50:45.624 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:50:45.633 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: writeCharacteristic
2025-07-07 15:50:45.646 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onCharacteristicWrite:
2025-07-07 15:50:45.647 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   chr: 5101
2025-07-07 15:50:45.647 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:50:46.004 20579-20579 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMethodCall: requestMtu
2025-07-07 15:50:46.024 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onMtuChanged:
2025-07-07 15:50:46.024 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   mtu: 512
2025-07-07 15:50:46.024 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: GATT_SUCCESS (0)
2025-07-07 15:50:58.022 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onConnectionStateChange:disconnected
2025-07-07 15:50:58.022 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: SUCCESS
2025-07-07 15:50:58.022 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] autoconnect is true. skipping gatt.close()
2025-07-07 15:50:58.026 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] onConnectionStateChange:disconnected
2025-07-07 15:50:58.026 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP]   status: SUCCESS
2025-07-07 15:50:58.026 20579-20603 [FBP-Android]           com.firo.mengqin                     D  [FBP] autoconnect is true. skipping gatt.close()
```

chipweinberger, 2025-07-09:
please open PR if you want it fixed. thanks.

chipweinberger, 2025-09-05:
yes youre right

chipweinberger, 2025-09-05:
fixed: dfb4d223b1cb8529ad85ed60aeda30d3637f11f9