chipweinberger/flutter_blue_plus #1248: Support Companion Device Manager

ZachHandley, 2025-08-27:
I quoted this issue just for context

Why in the world do you guys not support Companion Device Manager, even a little bit?

I've been trying to get code to work that would use the CDM as the "brain" -- the auto-connect, higher background processing priority, etc.

The problem is that (seemingly), CDM handles the bonding and pairing automatically, and then FBP tries to connect, and it fails. I don't get why this isn't supported, this seems like a pretty common usecase, imo.

> One way to do this would be to add an option to register the target BLE device as a Companion Device.
> 
> I think the problems with using Companion Device are
> 
> - The required callbacks are not supported until Android 12 or later.
> 
> - Pairing and bonding required to register as a Companion Device
> 
> 
> 
> https://developer.android.com/develop/connectivity/bluetooth/companion-device-pairing?#keep-awake
> 
> 
> 
> I'd like to hear your thoughts on adding Companion Device functionality to FBP. 

 _Originally posted by @cos-overclock in [#930](https://github.com/chipweinberger/flutter_blue_plus/issues/930#issuecomment-2508885909)_

ekuleshov, 2025-08-27:
@ZachHandley perhaps because FBP is a Bluetooth LE plugin and CDM is for Bluetooth Classic...

chipweinberger, 2025-08-28:
> Why in the world do you guys not support Companion Device Manager, even a little bit?

Lastly, this is a terrible way to get people to care about your issue or want to work with you.

chipweinberger, 2025-08-28:
Ya the more i read about CDM, the less I think it belongs in FBP. It should be a separate package.

closed 2025-08-28 by chipweinberger