chipweinberger/flutter_blue_plus #1257: [Help]: Write characteristics not working on iOS after updating to 1.35.7

jpzapata-clover, 2025-09-09:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.7

### Flutter Version

3.27.4 

### What OS?

iOS

### OS Version

18.6.2

### Bluetooth Module

Integrated

### What is your problem?

Writing to a characteristic of a service is not working after updating to 1.35.7 (or higher).

The following code previously worked but now it doesn't:

`var char = service.characteristics.firstWhereOrNull((c) => c.characteristicUuid.str == bleWriteIdentifier);
    if (char != null) {
      var newLine = String.fromCharCode(10);
      await char.write(
        utf8.encode(";ci ${exposure.ir}" + newLine),
        withoutResponse: false,
      );
      await char.write(
        utf8.encode(";cw ${exposure.white}" + newLine),
        withoutResponse: false,
      );
      await char.write(
        utf8.encode(";cu ${exposure.uv}" + newLine),
        withoutResponse: false,
      );
    } else {
      throw Exception("Exposure characteristic not found");
    }`

The characteristic is found on service.characteristics, so char variable is not null but upon calling the write method, an exception ocurrs: 

`"characteristic not found in service (chr: '6e400002-b5a3-f393-e0a9-e50e24dcca9e', svc: '6e400001-b5a3-f393-e0a9-e50e24dcca9e')"
`
Same exact code works fine on 1.35.5

### Logs

```shell
characteristic not found in service (chr: '6e400002-b5a3-f393-e0a9-e50e24dcca9e', svc: '6e400001-b5a3-f393-e0a9-e50e24dcca9e')
```

chipweinberger, 2025-09-09:
can you try to debug it and open a PR?

hollodotme, 2025-09-10:
Having the same issue and had to rollback to 1.35.5 and darwin 4.0.1. I had a look at [the changes since 1.35.5] in the darwin(https://github.com/chipweinberger/flutter_blue_plus/commits/master/packages/flutter_blue_plus_darwin), but I was not able to spot a significant change that causes the exception. The refactorings and improvements all look good.

Unfortunately, the changelogs are not up to date, making it somewhat difficult to follow the intentions behind the changes in each subpackage.

ekuleshov, 2025-09-10:
@jpzapata-clover @hollodotme I just tried the latest with my own apps and all seems to work on iOS (read, write, notifications)

Run `flutter pub deps` command and verify that you have the latest `flutter_blue_plus_darwin` plugin version.

You should NOT have have version locked for `flutter_blue_plus_darwin` in your `pubspec.yaml`.

Use `flutter pub upgrade` to get the latest versions of federated plugins. Just running `flutter pub get` after changing version of `flutter_blue_plus` is not enough.

You may also need to update CocoaPods after doing pub upgrade to get latest Darwin libs

```
flutter clean && flutter pub get
cd ios
pod repo update
pod update
pod install
```

[image]

hollodotme, 2025-09-10:
@ekuleshov Thanks for your feedback, just tried it again:

1. `pubspec.yaml`

```yaml
flutter_blue_plus: ^1.35.10
```

2. `flutter clean && flutter pub get`

[image]

3. `pod repo update`       
```bash
Updating spec repo `trunk`
```

4. `pod update`
```bash
Update all pods
Updating local specs repositories
Analyzing dependencies
Downloading dependencies
Generating Pods project
Integrating client project
Pod installation complete! There are 14 dependencies from the Podfile and 15 total pods installed.
```

5. `pod install`
```bash
Analyzing dependencies
Downloading dependencies
Generating Pods project
Integrating client project
Pod installation complete! There are 14 dependencies from the Podfile and 15 total pods installed.
```

6. `flutter pub deps`

[image]

7. `fvm list`

[image]

8. Starting the app and connecting the BT devices

Devices get connected, but as soon as I want to write to it, I get the following Error for each connected device:

```bash
[+6707 ms] [ERROR:flutter/runtime/dart_vm_initializer.cc(40)] Unhandled Exception: PlatformException(setNotifyValue, characteristic not found in service (chr: '6e400003-b5a3-f393-e0a9-e50e24dcca9e', svc: '6e400001-b5a3-f393-e0a9-e50e24dcca9e'), null, null)
[        ] #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
[        ] #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:367:18)
[        ] <asynchronous suspension>
[        ] #2      FlutterBluePlusDarwin._invokeMethod (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:285:17)
[        ] <asynchronous suspension>
[        ] #3      FlutterBluePlusDarwin.setNotifyValue (package:flutter_blue_plus_darwin/flutter_blue_plus_darwin.dart:213:12)
[        ] <asynchronous suspension>
[        ] #4      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:599:14)
[        ] <asynchronous suspension>
[        ] #5      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:283:11)
[        ] <asynchronous suspension>
```

I also see both characteristics (notify and write) while debugging:

[image]

However, when I roll back to `v1.35.5` and run `flutter pub get` without any other changes, everything works again as expected, and I can write to the devices.

TreyThomas93, 2025-09-10:
Can confirm downgrading to 1.35.5 worked for me.

chipweinberger, 2025-09-10:
The cause is most likely this change: 

https://github.com/chipweinberger/flutter_blue_plus/commit/62b1c78351bf69abf5169f114996bd35b5029850

For this issue: 

https://github.com/chipweinberger/flutter_blue_plus/issues/795

I don't have time to look into it at the moment, but feel free to open a PR.