chipweinberger/flutter_blue_plus #1263: [Bug]: broken in 1.36.2

ppjjzz, 2025-09-13:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.36.2

### Flutter Version

3.32.8

### What OS?

Android

### OS Version

android 14

### Bluetooth Module

pixel 9

### What is your problem?

when i updated to flutter_blue_plus-1.36.2, i can't build my app

### Logs

```shell
../../../.pub-cache/hosted/pub.dev/flutter_blue_plus-1.36.2/lib/src/bluetooth_descriptor.dart:32:24: Error: A value of type 'int?' can't be assigned to a variable of type 'int' because 'int?' is nullable and 'int' isn't.
bluetooth_descriptor.dart:32
        instanceId = p.instanceId;
                       ^
Target kernel_snapshot_program failed: Exception
2

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileFlutterBuildDebug'.
> Process 'command '/Users/panjiazhi/Documents/develop/flutter/bin/flutter'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 14s
Error: Gradle task assembleDebug failed with exit code 1
```

jhonny2pamp, 2025-09-13:
Not only in this version, changing to old versions this error persist

jhonny2pamp, 2025-09-13:
@chipweinberger your last commit have changes on instanceId, you believe that this have impact on this issue ?

chipweinberger, 2025-09-13:
will fix later

chipweinberger, 2025-09-13:
I'm not reproducing this issue, but try 1.36.3

make sure you run `flutter clean` too

vknow360, 2025-09-13:
> I'm not reproducing this issue, but try 1.36.3
> 
> make sure you run `flutter clean` too

Tried, same issue.

chipweinberger, 2025-09-13:
okay open a PR. for some reason its not happening for me

chipweinberger, 2025-09-13:
try 1.36.4

chipweinberger, 2025-09-13:
i hate all this federated plugin shit. will fix.

chipweinberger, 2025-09-13:
try again, 1.36.5

utkuvrs, 2025-09-13:
Previously I was using `flutter_blue_plus: ^1.32.12` changing it back to this version still gives the same issue as well, I hope it helps.

chipweinberger, 2025-09-13:
try again, 1.36.5

utkuvrs, 2025-09-13:
1. `flutter clean && flutter pub get`
2. Checked pubspec.lock
```
  flutter_blue_plus:
    dependency: "direct main"
    description:
      name: flutter_blue_plus
      sha256: "231104c1b8a19eda4d9f15d3904ecbb47256098ffb9dc1b51ab969f6ca373c48"
      url: "https://pub.dev"
    source: hosted
    version: "1.36.5"
```
3. Got this error:
```
C:\Users\xxx\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_blue_plus_android-7.0.2\android\src\main\java\com\lib\flutter_blue_plus\FlutterBluePlusPlugin.java:1660: error: ',', ')', or '[' expected
                                        String primaryServiceUuid
                                                                 ^
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus_android:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --info option to get more log output.
> Run with --scan to get full insights.

BUILD FAILED in 39s
Error: Gradle task assembleDebug failed with exit code 1

```

utkuvrs, 2025-09-13:
`√ Built build\app\outputs\flutter-apk\app-debug.apk`

Issue has been fixed with the `1.36.6` version

Thank you 👍

chipweinberger, 2025-09-13:
btw @utkuvrs , if you want the most stable version i recommend `1.34.5`. That's what I personally use, before we switched to federated.

Then 1.35.5 added new features that broke a bunch of stuff. I shouldn't have accepted that PR....

We should be approaching stable again though, after all that turmoil. 1.36.6 might be a version we stay on for awhile. TBD.

closed 2025-09-13 by chipweinberger

utkuvrs, 2025-09-13:
I've pinned the version to the stable version now.

I'd like to give some more feedback, it's working flawlessly with Zebra printers, however this federalized versioning doesn't really look like a good idea.

This is the first time I've experienced a versioning problem with this package, most of the apps don't need Linux/macOS/Windows support along with mobile platforms. I hope this helps. Have a good day.

utkuvrs, 2025-09-13:
I'm not exactly sure how they handle the versioning in this package but perhaps this might help:

https://pub.dev/packages/google_ml_kit

They also have a lot of packages under one package, perhaps this can be a good model to scaffold from.