chipweinberger/flutter_blue_plus #1266: [Help]: ERROR_GATT_WRITE_REQUEST_BUSY

topjx, 2025-09-16:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.35.5

### Flutter Version

3.32.8

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

I don't know, I don't know how to check

### What is your problem?

I have set up automatic reconnection. When frequently disconnecting and reconnecting, there is a high probability that this error will be reported when writing messages

ERROR_GATT_WRITE_REQUEST_BUSY

### Logs

```shell
FlutterBluePlusException | writeCharacteristic | fbp-code: 1 | Timed out after 15s
PlatformException(writeCharacteristic, gatt.writeCharacteristic() returned 201 : ERROR_GATT_WRITE_REQUEST_BUSY, null, null)
```

chipweinberger, 2025-09-16:
this error comes from Android, and I don't think FBP can fix it for you.

chipweinberger, 2025-09-30:
also - I just want to add I do *not* have this problem in my own code. But my app uses 1.34.5, which is a bit out of date.

sajath-45, 2025-10-07:
I use  flutter_blue_plus: ^1.36.2
Android version: 13
I have this same issue for only specific devices such as OnePlus but works with other android devices.
I tried many solutions such as

1. adding sufficient Delays Between Operations.
2. adding a Queue so that only ONE GATT operation happens at a time
3. Retry mechanism with exponential backoff: Up to 3 retries with increasing delays

But nothing worked.
This appears to be specific to OnePlus devices (particularly IN2025 model). 

Any guidance on:

1. OnePlus-specific workarounds or configurations
2. Native Android GATT queue management
3. Alternative approaches to handle GATT_WRITE_REQUEST_BUSY
4. Whether this is a known limitation with OnePlus's BLE implementation

Thank you for your assistance!

chipweinberger, 2025-11-05:
This seems like a bug in OnePlus's Android implementation. 

I'm going to close this until we have more evidence that there is something wrong in FBP.