chipweinberger/flutter_blue_plus #1271: [Help]: manufacturerData often missing on iOS but works fine on Android

MakeSomeFakeNews, 2025-09-19:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.36.8

### Flutter Version

3.32.8

### What OS?

iOS

### OS Version

18.6.2

### Bluetooth Module

esp32c3

### What is your problem?

Hi,

I'm having an issue with flutter_blue_plus where the manufacturerData field is frequently missing on iOS, while the same code works fine on Android.

Android version & device: (works fine)

Steps to Reproduce

Start scanning with flutterBluePlus.startScan().

Wait for scan results.

Observe that ScanResult.advertisementData.manufacturerData is often empty on iOS, but contains data on Android for the same peripheral.

Expected Behavior

manufacturerData should be available consistently on iOS just like on Android (if the peripheral is advertising it).

Actual Behavior

On iOS, manufacturerData is frequently missing or empty, even though Android reports it correctly.

```dart
scanSubscription = FlutterBluePlus.onScanResults.listen((results) {
        for (ScanResult result in results) {
          final device = result.device;
          final deviceId = device.remoteId.toString();

          if (!foundDeviceIds.contains(deviceId)) {
            foundDeviceIds.add(deviceId);

            final advertisementData = result.advertisementData;

            final serviceUuids = advertisementData.serviceUuids.map((uuid) => uuid.toString().toUpperCase()).toList();

            final manufacturerData = <int, List<int>>{};
            for (var entry in advertisementData.manufacturerData.entries) {
              manufacturerData[entry.key] = entry.value;
            }
             print('advertisementData=$manufacturerData---name=${device.platformName}');

            final rawAdvertisementData = <String, dynamic>{
              'advName': advertisementData.advName,
              'txPowerLevel': advertisementData.txPowerLevel,
              'serviceData': advertisementData.serviceData,
              'connectable': advertisementData.connectable,
            };

            if (device.platformName.isNotEmpty) {
              final deviceInfo = BluetoothDeviceInfo(
                address: deviceId,
                name: device.platformName.isNotEmpty ? device.platformName : (advertisementData.advName.isNotEmpty ? advertisementData.advName : 'Unknown Device'),
                serviceUuids: serviceUuids,
                rssi: result.rssi,
                manufacturerData: manufacturerData.isNotEmpty ? manufacturerData : null,
                rawAdvertisementData: rawAdvertisementData,
              );
              devices.add(deviceInfo);
            }
            // LoggerService.bluetooth('$_logTag: 发现设备: ${deviceInfo.name} (RSSI: ${deviceInfo.rssi}dBm)');
          }
        }
      });
```

### Logs

```shell
flutter: advertisementData={}---name=Sertorf <---- this device
flutter: advertisementData={1704: [1, 50, 50, 48, 52, 48, 48, 56, 55, 65, 67, 48, 51, 53, 56]}---name=midea
flutter: advertisementData={}---name=SpeedyBee WING-BLE
```

chipweinberger, 2025-09-19:
never seen an issue like this before, but I can't think of any cause for it that would be due to FBP

MakeSomeFakeNews, 2025-09-22:
```dart
  static Future<List<DeviceInfo>> scanDevices({Duration? timeout}) async {
    LoggerService.bluetooth('$_logTag: startScan');

    final completer = Completer<List<DeviceInfo>>();
    final devices = <BluetoothDeviceInfo>[];
    final foundDeviceIds = <String>{};
    StreamSubscription? scanSubscription;

    try {
      await FlutterBluePlus.adapterState.where((val) => val == BluetoothAdapterState.on).first.timeout(const Duration(seconds: 10));

      scanSubscription = FlutterBluePlus.onScanResults.listen((results) {
        for (ScanResult result in results) {
          final device = result.device;
          final deviceId = device.remoteId.toString();

          if (!foundDeviceIds.contains(deviceId)) {
            foundDeviceIds.add(deviceId);

            final advertisementData = result.advertisementData;

            final serviceUuids = advertisementData.serviceUuids.map((uuid) => uuid.toString().toUpperCase()).toList();

            final serviceData = <String, List<int>>{};
            for (var entry in advertisementData.serviceData.entries) {
              serviceData[entry.key.toString()] = entry.value;
            }

            LoggerService.bluetooth('$_logTag: [iOS]: ${result.advertisementData}, device: ${device.platformName}');

            final rawAdvertisementData = <String, dynamic>{
              'advName': advertisementData.advName,
              'txPowerLevel': advertisementData.txPowerLevel,
              'serviceData': serviceData,
              'connectable': advertisementData.connectable,
            };

            if (device.platformName.isNotEmpty) {
              final deviceInfo = BluetoothDeviceInfo(
                address: deviceId,
                name: device.platformName.isNotEmpty ? device.platformName : (advertisementData.advName.isNotEmpty ? advertisementData.advName : 'Unknown Device'),
                serviceUuids: serviceUuids,
                rssi: result.rssi,
                serviceData: serviceData.isNotEmpty ? serviceData : null,
                rawAdvertisementData: rawAdvertisementData,
              );
              devices.add(deviceInfo);
            }
            // LoggerService.bluetooth('$_logTag: device: ${deviceInfo.name} (RSSI: ${deviceInfo.rssi}dBm)');
          }
        }
      });

      FlutterBluePlus.cancelWhenScanComplete(scanSubscription);

      await FlutterBluePlus.startScan(timeout: timeout ?? const Duration(seconds: 3), continuousUpdates: true);

      await FlutterBluePlus.isScanning.where((val) => val == false).first;

      final sortedDevices = CwDeviceIdentifier.sortDevicesByPriority(devices);
      completer.complete(sortedDevices.cast<DeviceInfo>());
    } catch (e) {
      LoggerService.bluetooth('$_logTag: scan failed: $e');
      scanSubscription?.cancel();
      completer.completeError(
        DeviceConnectionException('Bluetooth dual service scan failed: $e'),
      );
    }

    return completer.future;
  }
```

zwells, 2025-09-26:
I'm wondering if something got broken here, I'm experiencing this as well after upgrading to FBP v2.0 from 1.35.5