chipweinberger/flutter_blue_plus #1272: [License]: the new FlutterBluePlus License (October 2025)

chipweinberger, 2025-09-21:
### Requirements

- [x] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

2.0.0

### Flutter Version

all

### What OS?

All

### OS Version

all

### Bluetooth Module

all

### What is your problem?

temp

### Logs

```shell
temp
```

Codel1417, 2025-09-21:
How would this affect apps purely developed by volunteers for a company? I develop an app that is published on their official accounts but other than that I am unaffiliated with them. I don't receive any compensation from the company nor do I want any.

kl3hm, 2025-09-24:
Dear Chip,

I am sitting here with mixed feelings. On the one hand, I understand that you want to generate profit from your own work. That is completely understandable and by no means reprehensible. On the other hand, I view your current approach somewhat critically, and I hope you have thought this through carefully. But let me explain further. 

This project could only become so big and successful because it is open source. Currently, 61 people have contributed to this project. They did so because they recognised the potential of this project and wanted to improve it, and certainly also to make their own work easier when developing a Flutter app. I may be wrong, but I find it hard to imagine that these people will continue to contribute to this project if they now have to pay for it. 

You want to capitalise on the whole thing now. As I said, that's your prerogative, but it leaves a bitter taste. Among other things, you use BlueZ from Canonical for example. What would you say if they suddenly wanted money for its use? Or Google for the general use of Flutter or a licence fee if you offer commercial libraries for Flutter?

As I said, I hope you have thought this step through carefully. But I find it hard to imagine that companies will pay $2999 a year for a single software package. In my opinion, it makes more sense in the long run for companies with more than 50 employees to develop their own Bluetooth library now.

Nevertheless, I wish you every success for the future.

liamcottle, 2025-09-24:
@chipweinberger I'd love to see the addition of GitHub Sponsors, or something along those lines, so those of us who don't have to purchase the commercial license can still contribute :) I had previously attempted to donate (without contacting you), but could not find any method listed in the project to do so.

kevinhikaruevans, 2025-09-25:
Couple of things:

- From the original BSD 3-clause license, you are required to retain the copyright notice when redistributed in binary form, i.e. you are required to display the original BSD-3 license within any app. Your license does not state this.
- Since some parts of the codebase are still licensed under BSD-3, you should include that in the LICENSE file, not in a separate file (as you effectively dual-licensing). The "View license" button and the link to the project's license on Github should navigate to the license of the codebase as a whole, not just to your relicensing.

chipweinberger, 2025-09-25:
> From the original BSD 3-clause license, you are required to retain the copyright notice when redistributed in binary form, i.e. you are required to display the original BSD-3 license within any app. Your license does not state this.

Thank you for raising this. To clarify:

- The new FBP License does not include that requirement because it is not a requirement of the FBP License, which only governs new releases going forward.
- The original BSD-3 license terms still apply to portions of the codebase contributed under BSD-3, and those notices remain in the repository under NOTICE.md.

Since those obligations are only tied to the BSD-licensed code, it would be misleading to include them in the text of the FBP License. Unfortunately, pub.dev only supports displaying a single LICENSE file, which makes it harder to make this relationship clear.

I agree that the relationship between the BSD-licensed code and the new FBP License should be clearer, so I’ll add a note in the preamble to explain this explicitly. That way, users know which obligations come from BSD-3 vs. FBP.

Importantly, I will not be enforcing the BSD-3 terms myself; so that relationship remains between users and the original BSD contributors.

> Personally, I think this move is pretty sleazy: to take a BSD-3 project, suffix it with "-plus" (normally for community forks), accept community contributions, only to relicense it to make a buck.

I understand why it might feel that way, but this is exactly what permissive licenses like BSD-3 are designed to allow. They explicitly permit modification, relicensing, and even commercial use.

It’s also worth noting that while there have been contributions from others, the overwhelming majority of the work—hundreds of hours over multiple years—has come from me. 

[image]

That means I’ve already given away years of work for free, which is something I think deserves recognition rather than being called “sleazy.” 

This is the balance BSD strikes: my version can move forward under a new license, and at the same time the original BSD-licensed code remains available to the community forever. 

If you or others would like to maintain a fork under BSD, you’re welcome to do so.

Shunt22, 2025-10-03:
Hi, @chipweinberger . 
I understand why do you want commercial license. I don't think this is a good idead though.

Anyway, speaking about commercial license, do you have a roadmap? What can we expect in further development?
This package has lack of windows support. Can we expect adding windows support in the future?

Feenposhleen, 2025-10-03:
First off, I sincerely appreciate the effort you put into this project. It's gone from beta-level to solid under your watch. So I understand why you'd want to be compensated for your efforts going forward, now that your own product is completed. While I don't agree with this particular way of doing it, it's your choice, and I wish you luck.

However, I think you should seriously reconsider the pricing. It is incredibly steep. I get that it's enticing to price it at a level where only a few buyers can sustain you, but it's sending out patent-troll vibes. 

In a lot of countries, $30k is a yearly salary. It might be small potatoes in a SV context, but you are essentially turning the use of your package into a non-starter for most of the world.

Perhaps you don't care about legacy, but there will be a fork. This fork will become the go-to BLE package for Flutter. FBP will be considered trap ware.

Are you fine with that?

bcliang, 2025-10-03:
@Feenposhleen I don't love the new license approach either, but the commercial pricing is $3k/year, not $30k. That seems to be reasonable pricing for medium-large businesses? I agree that a new open fork will show up sooner or later (maybe it'll be the l2cap fork mentioned earlier). It remains to be seen whether license terms will result in user migration away from FBP. For me personally, package support/stability is more important (while we're complaining about things.. why do the platform interfaces follow semver while the main project uses a random versioning strategy??!?)

The benefit to a commercial license is that @chipweinberger should start to be more developer-friendly -- open to i.e. feature requests that he doesn't use/need. I echo @Shunt22's comments that a roadmap for the project would be nice.

chipweinberger, 2025-10-03:
The current roadmap is empty i.e. business as usual fixing any issues that pop up, supporting new versions of iOS, Android, etc.

For larger features i'll wait until there is commercial interest in that feature. 

If you want a new feature without buying a license, the best way is through PRs. 

So far I have not liked the L2Cap PRs. They have not followed the same style as the existing code base. i.e. minimal class use, single file, etc.

chipweinberger, 2025-10-19:
@ImanGhasemiArani There is no key. The "license" argument in the connect function has no effect. You can just look at the code. The license argument is unused.

It's just there so you  know about the new license change if you upgrade to 2.X.